adafruit/Adafruit-SSD1331-OLED-Driver-Library-for-Arduino #7: ESP8266 compilation error on nodemcu v3

Netoperz, 2016-11-30:
Using Arduino ide 1.6.12 
and esp8266 node mcu v3

i've set the pins according to gpios of the nodemcu v3:

// Color definitions
#define  BLACK           0x0000
#define BLUE            0x001F
#define RED             0xF800
#define GREEN           0x07E0
#define CYAN            0x07FF
#define MAGENTA         0xF81F
#define YELLOW          0xFFE0  
#define WHITE           0xFFFF

#include <Adafruit_GFX.h>
#include <Adafruit_SSD1331.h>
#include <SPI.h>

// You can use any (4 or) 5 pins 
#define sclk 14
#define mosi 13
#define cs   15
#define rst  12
#define dc   4

and while compiling the example i got this:

/Users/dev/Documents/Arduino/libraries/Adafruit_SSD1331_OLED_Driver_Library_for_Arduino/Adafruit_SSD1331.cpp: In member function 'void Adafruit_SSD1331::begin()':
/Users/dev/Documents/Arduino/libraries/Adafruit_SSD1331_OLED_Driver_Library_for_Arduino/Adafruit_SSD1331.cpp:289:21: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment
         sclkportreg = portOutputRegister(digitalPinToPort(_sclk));
                     ^
/Users/dev/Documents/Arduino/libraries/Adafruit_SSD1331_OLED_Driver_Library_for_Arduino/Adafruit_SSD1331.cpp:293:20: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment
         sidportreg = portOutputRegister(digitalPinToPort(_sid));
                    ^
/Users/dev/Documents/Arduino/libraries/Adafruit_SSD1331_OLED_Driver_Library_for_Arduino/Adafruit_SSD1331.cpp:305:15: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment
     csportreg = portOutputRegister(digitalPinToPort(_cs));
               ^
/Users/dev/Documents/Arduino/libraries/Adafruit_SSD1331_OLED_Driver_Library_for_Arduino/Adafruit_SSD1331.cpp:308:15: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment
     rsportreg = portOutputRegister(digitalPinToPort(_rs));
               ^
exit status 1
Błąd kompilacji dla płytki NodeMCU 1.0 (ESP-12E Module).

Netoperz, 2016-11-30:
Ok using fix from @tectiv3	ESP8266 compilation error fix
makes compilation possible but the OLED is not responding.

It migt be pin configuration problem.

ladyada, 2018-06-19:
sorry for delay on this BUT i think we fixed this in the "spitft" branch - can you try that and let me know?

closed 2018-06-19 by ladyada