adafruit/Adafruit-SSD1331-OLED-Driver-Library-for-Arduino #9: ESP8266 "PortReg" type errors - Fixed now.

artkeller, 2017-01-21:
Dear Adafruit Team! 
Thanks for the *Adafruit-SSD1331-OLED-Driver-Library-for-Arduino* lib.

I compiled with ESP8266 but got *PortReg* type errors. 
The following fix in file *Adafruit_SSD1331.h* worked fine for me.

```
line 25:
#ifdef __SAM3X8E__
typedef volatile RwReg PortReg;
typedef uint32_t PortMask;
#define _BV(b) (1<<(b))
#elif defined(ESP8266)
typedef volatile uint32_t PortReg;
typedef uint32_t PortMask;
#else
typedef volatile uint8_t PortReg;
typedef uint8_t PortMask;
#endif
```

- Board:  ESP8266
- Arduino IDE 1.6.12

BR, Thomas

mattratto, 2018-05-22:
did same for feather nrf52 - fixed issues.

ladyada, 2018-06-19:
hiya i think we fixed this in the "spitft" branch - can you try that and let me know?

closed 2018-06-19 by ladyada