adafruit/Adafruit-SSD1331-OLED-Driver-Library-for-Arduino #18: SSD1331 1.1.0 makes writeCommand() inaccessible (also can't be reverted)

Paraphraser, 2018-07-17:
- Arduino board:  WeMos D1 R2 & mini 

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.5

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

I have a physical button in my project which, when pressed, results in a call to one of the following to toggle the on/off state of the OLED:

`    display.writeCommand(SSD1331_CMD_DISPLAYOFF);`
or
`    display.writeCommand(SSD1331_CMD_DISPLAYON);`

This gives me the ability to only have the OLED turned on when I need to interact with the project, hopefully also extending the life of the OLED and minimising any burn-in.

The upgrade to GFX 1.2.5 and SSD1331 1.1.0 breaks this code. Although the _CMD_ constants are still visible in the SSD1331 header file, there is no longer a way to call writeCommand() from a sketch.

Also, when I reverted to the library versions I had installed previously (GFX 1.2.3, SSD1331 1.0.2) my sketch no longer compiles, reporting:

`Adafruit_SSD1331_OLED_Driver_Library_for_Arduino/Adafruit_SSD1331.cpp:289:21:`
` error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment`
`         sclkportreg = portOutputRegister(digitalPinToPort(_sclk));`
`                     ^`
`Adafruit_SSD1331_OLED_Driver_Library_for_Arduino/Adafruit_SSD1331.cpp:293:20:`
` error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment`
`         sidportreg = portOutputRegister(digitalPinToPort(_sid));`
`                    ^`
`Adafruit_SSD1331_OLED_Driver_Library_for_Arduino/Adafruit_SSD1331.cpp:305:15:`
` error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment`
`     csportreg = portOutputRegister(digitalPinToPort(_cs));`
`               ^`
`Adafruit_SSD1331_OLED_Driver_Library_for_Arduino/Adafruit_SSD1331.cpp:308:15:`
` error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment`
`     rsportreg = portOutputRegister(digitalPinToPort(_rs));`
`               ^`
`[snip]`
`Using library Adafruit_GFX_Library at version 1.2.3 in folder: ~/Documents/Arduino/libraries/Adafruit_GFX_Library`
`Using library Adafruit_SSD1331_OLED_Driver_Library_for_Arduino at version 1.0.2 in folder: ~/Documents/Arduino/libraries/Adafruit_SSD1331_OLED_Driver_Library_for_Arduino`
`[snip]`
`Error compiling for board WeMos D1 R2 & mini.`

I have two computers which I (mostly) keep in parallel. The computer not yet upgraded to GFX 1.2.5 and SSD1331 1.1.0 still compiles the sketch properly so I am reasonably certain that upgrading the other computer broke something that downgrading is unable to repair.

I tried deleting both the GFX and SSD1331 libraries completely, then reinstalling GFX 1.2.3 and SSD1331 1.0.2 from scratch but that did not cure the problem.

ladyada, 2018-07-17:
yeah we now subclass SPITFT which means our lives are not doomed. we could make writecommand public but because writeCommand() didnt start an SPI transaction, its a dangerous command to run :)

would you be interested in adding a new command to SPITFT called like enableDisplay(bool flag) just like invertDisplay?
https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_SPITFT.cpp#L481
we'd take the PR and bump both libraries

Paraphraser, 2018-07-17:
a function like enableDisplay() would certainly solve my problem so this approach has my vote.

ladyada, 2018-07-17:
@Paraphraser OK! we have no plans or eta on adding this function so if you're willing to try forking and implementing it, we'd be willing to add it!

ladyada, 2019-04-18:
we now have sendCommand() which does all your SPI transaction stuff for you, please use that if you plan to still add this!

Paraphraser, 2019-04-21:
Your last advice about sendCommand() made me think that it really was about time I got around to doing this. You’ll find the PR in your queue. Thanks for the kick-in-the-pants.

closed 2019-04-21 by Paraphraser