adafruit/Adafruit-SSD1331-OLED-Driver-Library-for-Arduino #23: Compile Error: 'sendCommand' was not declared in this scope

PowerBroker2, 2019-04-26:
Trying to run test.ino and got the compile-time error in issue title. The libraries I have installed for the test case is Adafruit_GFX and Adafruit_SSD1331. Is sendCommand supposed to be declared in another library I'm missing or was sendCommand accidentally not included in the master branch?

makermelissa, 2019-04-26:
It should be in the latest version of Adafruit GFX.

PowerBroker2, 2019-04-26:
I see that it is in Adafruit_SPITFT within the GFX library, but neither test.ino nor Adafruit_GFX.h include Adafruit_SPITFT.h, thus causing the compile-time error...

makermelissa, 2019-04-26:
That's odd. It should be on lines 190 and 191 of Adafruit_SPITFT.h file.

makermelissa, 2019-04-26:
Which version of Adafruit_GFX are you running?

makermelissa, 2019-04-26:
See https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_SPITFT.h#L190

PowerBroker2, 2019-04-26:
As per my last comment, sendCommand is in Adafruit_SPITFT.h but that particular header isn't included anywhere for test.ino. I was able to fix test.ino by adding Adafruit_SPITFT.h to the include section of the sketch:

`
#include <Adafruit_SPITFT.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1331.h>
#include <SPI.h>
`

makermelissa, 2019-04-26:
That's because it's included in Adafruit_SSD1331.h which is included in the sketch. See https://github.com/adafruit/Adafruit-SSD1331-OLED-Driver-Library-for-Arduino/blob/master/Adafruit_SSD1331.h#L17

PowerBroker2, 2019-04-26:
Ah I see, but the problem still persists.

I did a clean install of the latest versions of both the GFX and SSD1331 libraries with no luck. I'm still getting the error. I installed both of the libraries using the Arduino Libraries Manager.

Any ideas what's going wrong?

makermelissa, 2019-04-26:
I'll try doing a clean install of both to see if I can reproduce it. So far I haven't been able to.

makermelissa, 2019-04-26:
Which board are you compiling for?

PowerBroker2, 2019-04-26:
Teensy 3.5

I rolled back to commit 53aaf5755b7a0bc237f92a500743c75cbe8af421 (Apr 5, 2019) and it works for me. Not sure why the current version doesn't.

makermelissa, 2019-04-26:
Ok, I'll try that one...

makermelissa, 2019-04-26:
Oh, did you install the teensyduino libraries? When I installed it I didn't install the included libraries and they may be using an older version of the SSD1331 library.

makermelissa, 2019-04-26:
It also might be their included version of the Adafruit GFX library causing issues. sendCommand was implemented in the last couple weeks.

PowerBroker2, 2019-04-26:
Yes, but I'm not sure if this is the core problem considering the fact that I also tried compiling for Uno but still got the same error. Maybe it would work better to include with parenthesis so that it tries to reference the user defined libs as installed by the libraries manager (instead of the possibly outdated libs from the core directory)? I'm not sure if that'll fix it, but might be something to test

makermelissa, 2019-04-26:
Ok, we'll have to play around with that. However the libraries published on github have been tested with the Teensy boards, so including it shouldn't be necessary. It's possible that the teensy package needs the updated library. Anyway, thank you for finding this. I'll leave the issue open for now. As a workaround for now, I would recommend removing the teensy included libraries and installing off the Arduino Library manager.

PowerBroker2, 2019-04-26:
Cool, sounds good. Tbh I was running Arduino 1.8.8 since the last time I looked, Teensyduino didn't support the latest version of the Arduino IDE. Just checked and they now have Teensyduino supported for Arduino 1.8.9. I'll just install the latest versions of both then install the latest vesions of the GFX and SSD1331 libraries through the library manager and see if I still get the error.

makermelissa, 2019-04-26:
Yeah, I ran into the same. I found a beta version that I was running. I just checked a few minutes ago as a result of this issue and noticed it had been released. :)

PowerBroker2, 2019-04-26:
Complete new install of Arduino and Teensyduino still gives the error. Looks like Teensyduino's libraries are still out of date and are causing the problem :(

makermelissa, 2019-04-26:
Ok, I was able to reproduce by installing the Adafruit_GFX library with the Teensyduino installer. In the area where the error messages show, it should say something like:
```
Multiple libraries were found for "Adafruit_GFX.h"
 Used: /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/Adafruit_GFX
 Not used: /Users/melissa/Documents/Arduino/libraries/Adafruit_GFX_Library
Error compiling for board Teensy 3.5.
```

I would recommend removing the folder in the location where it used it.

ladyada, 2019-04-26:
@PaulStoffregen hiya since we are doing a lot of updates to GFX and our displays,  please point people to the Arduino library manager to update/install libraries instead of bundling

makermelissa, 2019-05-16:
Nothing left to do on this here.

closed 2019-05-16 by makermelissa